wavelog/wavelog #1674: button style is different

hellofinch, 2025-02-19:
**Describe the bug**
As shown in the image, the style of the CSV button is inconsistent with the other buttons. I'm not sure if this is a display bug, a setting in Wavelog, or simply a case of not applying a uniform style. I believe these buttons should have a consistent appearance.

**Version Information**
- OS:               linux
- PHP:             8.3.13
- MySQL/MariaDB:    10.6.18-MariaDB-0ubuntu0.22.04.1
- Wavelog:          dev-[2f19bb9d](https://github.com//Wavelog/commit/2f19bb9df969878589fb5088a04f71940cb2436e)

**Steps to Reproduce**
just display error

**Expected Behavior**
these buttons should have a consistent style.

**Screenshots**

[image]

AndreasK79, 2025-02-19:
It should be fixed in the dev branch now. Hopefully I found all places.

hellofinch, 2025-02-20:
COOL! Thank you!
73
: )

closed 2025-02-20 by hellofinch