wavelog/wavelog #1679: SN100IARU generates wrong QRZ link

lukezaa, 2025-02-19:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
SN100IARU generates wrong link do QRZ.com

https://www.qrz.com/db/SN10%C3%98IARU

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:  Ubuntu 24.04
- PHP:  8.3
- MySQL/MariaDB:  MySQL 8.0.41
- Wavelog:  2.0

**Steps to Reproduce**
1. Try to log QSO with SN100IARU
2. Click QRZ

**Expected Behavior**
Redirect to SN100IARU not SN10ØIARU

**Screenshots**
[image]

**Desktop Information:**
- OS:  Ubuntu 24.04
- Browser: Firefox
- Version:  135

phl0, 2025-02-19:
Good find. Fix in https://github.com/wavelog/wavelog/pull/1680

phl0, 2025-02-19:
Fixed in dev branch now. Will be part of the next release.

closed 2025-03-01 by github-actions[bot]