wavelog/wavelog #1681: Automatic Docker Builds on GitHub Actions

ethancedwards8, 2025-02-20:
**What Feature is Missing?**
Automatic docker builds for Wavelog dev and master branches

**Why is This Feature Important?**
I run the dev branch in my homelab and would appreciate it if automated Docker builds for the dev branch (and the main branch too) were activated. I run the dev branch as I like to have the newest features but I also like to be a tester to see if any bugs pop up so I can report them. Currently, I have to build the image locally and deploy it on my server. Y'all already have a GHA (and GHA is free!) workflow for it, it just needs to be enabled to be ran on push/merge. 

I recognize that y'all are using ghcr.io and you probably have to pay for it. Instead, y'all could use https://hub.docker.com/, which is free for public images.

**Is This Feature Personal or Beneficial to Others as Well?**
It will help others who want to run the dev branch on docker.

int2001, 2025-02-20:
The master-image is automatically built up on release. there's an action for it since 2024. So if you use the tag `:latest` in your docker-compose you'll receive the latest master.

The dev-image is built on demand by us, to save ressources. we're not planning to change this

ethancedwards8, 2025-02-20:
Got it

closed 2025-02-20 by ethancedwards8