wavelog/wavelog #1715: Incorrect warning "active station location isn't linked to your logbook"

R1BLH, 2025-02-28:
User with access level "Operator" has the logbook with linked station location. But on the dashboard WL shows warning: "Your active Station Location isn't linked to your Logbook"

[image]

[image]

[image]

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
Version 	2.0.1 Latest Version
Language 	English
Base URL 	https://wavelog.cats-home.net/
Migration 	236
Environment 	production
System Time 	28.02.2025 11:09:41
	
Total QSO on this instance 	130,775 QSOs
Total User 	10 Users

Server Software 	Apache/2.4.52 (Ubuntu)
PHP Version 	8.2.27 OK
MySQL Version 	11.3.2-MariaDB-1:11.3.2+maria~ubu2004
Codeigniter Version 	3.2.0-dev

HB9HIL, 2025-02-28:
I assume that this happens because the currently active station location is not linked with the currently active station logbook. Therefor I'm not 100% if this is a bug or just "works like designed". Probably @AndreasK79 can give a much better opinion here.

R1BLH, 2025-02-28:
But the currently active station location is linked to the currently active station logbook!

Please look at the screenshots. 
The location with id 100 has the green mark in the list, that means it is linked to the active logbook.
The second point is the number of QSOs: this location contains 14710 QSOs. On the screenshot of the dashboard QSO breakdown shows 14710 QSO. And the QSO table is not empty, neither on the dashboard nor on the logbook views (Overview and Advanced).

HB9HIL, 2025-02-28:
[image]

Yes but not the ACTIVE station **location** is linked.. or better.. The linked location is not the active one

HB9HIL, 2025-02-28:
Therefore the warning is atleast not "wrong". Since the current active location (the first one) is not linked

HB9HIL, 2025-02-28:
The warning says "The **active** Station Location isn't linked..."

R1BLH, 2025-02-28:
> The warning says "The **active** Station Location isn't linked..."

Mea culpa... Where were my eyes? How could I not notice? 
My apologies for the inconvenience.

HB9HIL, 2025-02-28:
No problem :) We are here to help :)