wavelog/wavelog #1719: DARC Contest Hub won't accept Wavelog EDI files

xyz667, 2025-03-02:
**Describe the bug**
The DARC Contest Hub does not accept Wavelog REG1TEST EDI files. I am pretty sure this used to work last time I tried (last 2024 VHF contest??).
I could resolve the error and make the EDI uploadable by removing the last semicolon in every line of the logged QSOs.

**Version Information**
Wavelog: 2.0.1

**Steps to Reproduce**
1. Log DARC VHF Contest in Wavelog.
2. Export EDI file.
3. Upload EDI to DARC Contest Hub.
4. Contest Hub complains about wrong log format (sorry exact wording not available anymore).

HB9HIL, 2025-03-02:
@DB4SCW

DB4SCW, 2025-03-02:
@xyz667  Please provide a sample file from your end (just for safety to rule out any data entry issues).

According to the REG1TEST definition, the last field of each QSO (starting with the "last semicolon") is for duplicate QSOs. Since there is no way in Wavelog (as of yet) to mark QSOs as duplicate, this is always empty. But it still is mandatory according to the format description.

Since the last 2024 contest, we did not change anything about the QSO lines, so a change on DARCs side seems to be the most obvious answer.

xyz667, 2025-03-03:
> [@xyz667](https://github.com/xyz667) Please provide a sample file from your end (just for safety to rule out any data entry issues, such as rogue semicolon).
> 
> According to the REG1TEST definition, the last field of each QSO (starting with the "last semicolon") is for duplicate QSOs. Since there is no way in Wavelog (as of yet) to mark QSOs as duplicate, this is always empty. But it still is mandatory according to the format description.
> 
> Since the last 2024 contest, we did not change anything about the QSO lines, so a change on DARCs side seems to be the most obvious answer - but let's see your data first.

Here is a sample EDI line for a QSO, from the Wavelog export:
`250301;1425;xxxx;1;59;001;59;039;;JOxxxx;121;N;N;N;;`

The DARC Contest Hub complains:

`Fehler in Zeile  #xx: Formatfehler in Logzeilen.
Bitte überprüfen, ob alle Inhalte wie Calls und Rapporte enthalten und korrekt geschrieben sind [15/dupe]:\ >>>250301;1425;xxxx;1;59;001;59;039;;JOxxxx;121;N;N;N;;<<<
Hilfe bei Fehler:
REG1TEST (= EDI)-Beispiel:141102;1023;DJ5MW;;599;179;599;042;;JN47WN;156;;;;`

Note "Hilfe bei Fehler" where they link to the EDI specs and a sample EDI entry.
The example has a different number of data fields as the Wavelog EDI.
As mentioned above, if I remove the last ";" in every QSO line there is no error and the file uploads fine.

DB4SCW, 2025-03-03:
Fix is available in PR #1729 

Removed the trailing semicolon for broader compatibility.

closed 2025-03-16 by github-actions[bot]