wavelog/wavelog #1721: Heading in Advanced page doesnt scroll with data

grinchdubs, 2025-03-03:
**Describe the bug**
It seems in the most recent update that when scrolling left and right that the heading row doesn't scroll with the data

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:    Kali latest version
- Server Software 	Apache/2.4.62 (Debian)
- PHP Version 	8.2.24 OK
- MySQL Version 	11.4.3-MariaDB-1
- Codeigniter Version 	3.2.0-dev
- Wavelog:  2.0.1

**Steps to Reproduce**
1. Go to Advanced page
2.use the scroll bar at the bottom to scroll right to left

**Expected Behavior**
Header should also scroll

**Screenshots**

[image]

phl0, 2025-03-03:
Cannot confirm. Scrolls correctly on my view. What operating system and browser are you using?

[image]
(Screenshot scrolled right)

grinchdubs, 2025-03-13:
Closing this as it seems like it is no longer an issue on my machine, not sure what happened/changed

closed 2025-03-13 by grinchdubs