wavelog/wavelog #1749: Contest Log: Band information not set by CAT interface

205er, 2025-03-09:
Hi,

I use Wavelog-Gate to gather information from my rig such as Frequency, Power or Mode.
Today I discovered, that in the "Live Contest-Logging" all the RIG's data are taken over correctly, BUT the BAND information. You always have to adjust BAND field manually. In "Live QSO" log, the band information is adjusted automatically in regard of the frequency.

[image]

Thanks & br

DJ3CE, 2025-03-12:
Hi, had a short look into this. This is a side-effect of a already fixed issue with a (missing) variable being referenced in an 'change'-callback which is also fired by the cat update.

So imho already fixed in the dev-Branch (see Commit 14956d828b).

205er, 2025-03-15:
thanks for the feedback

closed 2025-03-15 by 205er