wavelog/wavelog #1767: eSQL did not work

ds2k5, 2025-03-14:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**

eSQL Import/Download fail - remove use and password from config

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Debian 12
- PHP:              8.3.16
- MySQL/MariaDB:    MySQL 8.0.37
- Wavelog:          Release 2.0.1 

**Steps to Reproduce**
1. Go to  -> User account
2. Click on eSQL enter User and PW ans safe
3. Go to 3party Services -> eSQL -> Upload
4. See error: You have not defined your eQSL.cc credentials!

**Expected Behavior**
Describe what you expected to happen.

**Screenshots**
Include screenshots if applicable to illustrate the issue.

[image]

**Desktop Information:**
- OS:               Linux Debian 12
- Browser:       Firefox
- Version:        128.8.0esr

**Additional Context**
Provide any other relevant context about the problem.

int2001, 2025-03-15:
That's not a bug.
eQSL-access relies on three things:
- Username
- Password
- Station-Nickname (set it at stationlocation)

Furthermore:
Wavelog automaticly removes User/Passwort from your Account if eQSL tells Wavelog "Wrong credentials".
Currently there's no "Message-Center" or sth. else where we can show this. But: if you enter your credentials (and the eQSL-Nickname) and manually(!) sync, you can see if everything is correct.

Please avoid special-characters for the eQSL-Password. eQSL is very very old and legacy. it has problems with some characters.

ds2k5, 2025-03-15:
OK 

I changed from email to username
but upload did fail
download worked

int2001, 2025-03-16:
please double-check the Nickname set at station-location to the Nickname which is set at eQSL.
They MUST fit. Furthermore the valid timerange at eQSL MUST fit.

closed 2025-03-30 by github-actions[bot]

ds2k5, 2025-03-30:
since update to 2.0.2 it works fine

ds2k5, 2025-04-17:
since Update to 2.0.3 stopped working

in station setup is  nickname set
username and passwort set as using to login on esql.cc

are there character that do not work in the password ?

int2001, 2025-04-18:
There's nothing known. eQSL works on several instances with WL 2.0.3. without any problems.

If you want to be sure:
- Avoid things like ?&$%^\'" - but tbh: works here even which such characters.
- double check your subscription at eQSL. especially the Call you want to upload (prefixes/suffixes of your own call)
- double check the validity of your "QTH" at eQSL and the Nickname

ds2k5, 2025-04-20:
@int2001 

thanks

>double check the validity of your "QTH" at eQSL and the Nickname

Is the same es in eqsl.cc Profile

>Avoid things like ?&$%^'" - but tbh: works here even which such characters.

Tryed a password without this characters - but did not work for me

Just wanna download the QSL-Cards from eqsql.... did not have an pro account there
so no subscription

int2001, 2025-04-21:
Sorry, can't help more.
We are not able to reproduce your problem. Everything works well on all instances i know of.

One more thing: do you use your email-address or Callsign as login for eQSL?