wavelog/wavelog #1771: No warning when LoTW sync fails after credentials are changed

dl3jop, 2025-03-16:
**Describe the bug**
Hey,
I recently changed my LoTW credentials and forgot to update them in wavelog.
Therefore the LoTW syncs obviously failed. But there is no indication for the user that the credentials are wrong or the login has failed.
Doing a "Manual Sync" in the LoTW view yields no error. In comparison, I tried the "LoTW Import" function which told me that my password is wrong.
In hindsight I guess the user knows that the password was changed and changes it in wavelog as well but in my case I simply forgot to do exactly that and went hunting in the logs to see why my LoTW QSLs were not syncing :D

**Version Information*+
- PHP:              [e.g. 8.1.30]
- MySQL/MariaDB:    [e.g. 10.11.6]
- Wavelog:          [e.g. 2.0.1, latest commit: da342c7b] 

**Steps to Reproduce**
1. Change LoTW password on lotw.arrl.org
2. Goto LoTW View in Wavelog
3. Initiate Manual Sync

Note: I've had no new QSO after I changed the password, so there were no new QSOs to be uploaded to LoTW.

int2001, 2025-03-17:
Hi Joshua,
that's normal. One must understand how LoTW works here:
- For Downloading the report, the password is needed. Everything works here (message "pw wrong")
- For Uploading LoTW needs **no password**. It's completely based on the cert-trust. Therefore no error here.

What we can think about:
since "Manual sync" does Up- and download together, we could show at least that the Download failed.

@phl0 pse correct if i'm wrong.

73, DJ7NT

phl0, 2025-03-19:
> [@phl0](https://github.com/phl0) pse correct if i'm wrong.

This is correct. Upload does not include authentication via username/password. Upload solely depends on a (valid) LoTW certificate. Files with missing or invalid signature(s) are simply ignored on LoTW side I guess.

The download function has an error handler that shows the issue with credentials. I just set my login name to bogus value and upon manual synchronization I get:

[image]

phl0, 2025-03-19:
This btw is the same error message as in manual LoTW download. So I should (have been) there upon your failed sync. Just at the bottom of the page below the upload infos.

closed 2025-03-31 by dl3jop

phl0, 2025-03-31:
Never mind. Could as well have been a nasty bug. We're here to help :-)