wavelog/wavelog #1773: Unable to QSO some callsign

remy56k, 2025-03-17:
Hi, 

I'm unable to post a manual QSO with some callsigns (Belgium, IARU centenary event).

Firefox version : 136.0.1 (64 bits)
Wavelog version : Release 2.0.1

When typed on keyboard or pasted from a string: 

[image]

If I click on "Search DXCluster for latest spot", the callsign is found and can be saved : 

[image]

Thanks,

int2001, 2025-03-18:
do you click "instant" to "save" after filling in the Call?

remy56k, 2025-03-18:
No, I fill the callsign during the contact, and just adjust the time at the end of the QSO. I know sometimes the DXCC search can take few seconds to be complete.

int2001, 2025-03-19:
okay. tbh: we were not able to reproduce the behaviour for OS100U (or other calls).
What does the network-Tab (at Browser-Debug) show, when you enter the call and change focus to another field?

And additional information: On what hardware are you hosting Wavelog?

phl0, 2025-03-19:
I did a bit of research:
- Empty DXCC tables cannot be the reason. Results would give NONE as DXCC in that case
- No lookup configured is also not the reason here because then the internal lookup function would give Belgium as DXCC and in case of empty tables just NONE as described before

remy56k, 2025-03-19:
The problem seems to be gone with OS100U since this morning 🥹
Callsigns are very new (special event since 00:00 UTC March 15, maybe this can explain something. 

More info on my hardware : DELL R640 on SSD DC-class, VM with 4G RAM and 4 vcores.

Thanks!