wavelog/wavelog #1775: Satellite Flightpath map not display

hellofinch, 2025-03-20:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
map not show in Satellite Flightpath page.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               linux
- PHP:              8.3.13
- MySQL/MariaDB:    10.6.18-MariaDB-0ubuntu0.22.04.1
- Wavelog:         dev-35333830

**Steps to Reproduce**
1. Go to 'Satellite Flightpath'
2. See error

**Expected Behavior**
The map may show below not blank. If the map works as designed, please let me know. : )

**Screenshots**
Include screenshots if applicable to illustrate the issue.

[image]

**Additional Context**
I'm not sure that my network has trouble. But the main page has a map too and it shows fine.

: )

AndreasK79, 2025-03-20:
@hellofinch seems I left the hardcoded tileurl in the code. I think #1776 should fix it. Can you test and confirm?

hellofinch, 2025-03-20:
Thanks for your reply.
I pull the latest commit. But the map is still missing.

As you can see, I think the red space has some problems,too.
[image]

[image]

the map in the dashboard works fine.
[image]

Have a nice day.
 : )

AndreasK79, 2025-03-20:
@hellofinch this was not merged into dev. I merged the PR. Try to fetch latest dev now :)

hellofinch, 2025-03-21:
Cool ! It works! Thanks!

73!
: )

closed 2025-03-21 by hellofinch