wavelog/wavelog #1782: Simple Fast Log Entry (SFLE) fails with error

DC1TC, 2025-03-21:
**Describe the bug**
Entering basic QSO info in Simple Fast Log Entry fails with a JavaScript error in the console.

**Version Information**
- Wavelog:          WaveLog 2.0.1
(Draussenfunker instance)

**Steps to Reproduce**
1. Go to QSO > Simple Fast Log Entry
2. Select a Station Location and Operator
3. Enter basic QSO data in the FLE syntax:
```
2m FM
1524 N0CALL 59 59
```

4. Press return
It does not show the QSO in the table on the right, revealing the console shows an JS error (see below).

**Expected Behavior**
QSO info appears in the table on the right

**Desktop Information:**
- OS:               macOS
- Browser:          Safari
- Version:          Version 18.2 (19620.1.16.111.6, 19620)

**Additional Context**
```
[Error] TypeError: addInfoMatches.forEach is not a function. (In 'addInfoMatches.forEach((item) => {
			row = row.replace(item[0], "");
			let kv;
			if (item[0][0] == '<' && (kv = item[1].match(/^([a-z_]+): *(.*)$/))) {
				add_info[kv[1]] = kv[2];
			} else if (item[0][0] == '[') {
				add_info.qslmsg = item[2];
			} else {
				add_info.comment = (('comment' in add_info)?add_info.comment+' ': '')+item[1];
			}
		})', 'addInfoMatches.forEach' is undefined)
	(anonymous function) (simplefle.js:250)
	forEach
	handleInput (simplefle.js:239)
	(anonymous function) (simplefle.js:595)
	dispatch (jquery-3.3.1.min.js:2:41778)
```

DC1TC, 2025-03-21:
Likely introduced with https://github.com/wavelog/wavelog/pull/1483?

HB9HIL, 2025-03-22:
@DJ3CE

DJ3CE, 2025-03-22:
Fixing Issue #1782 with PR #1783 ;). Thanks for reaching out, looks like Safari is a bit [behind all the others](https://developer.mozilla.org/de/docs/Web/JavaScript/Reference/Global_Objects/Iterator/forEach)…

@HB9HIL, please double check :).

DC1TC, 2025-03-24:
Fixed with https://github.com/wavelog/wavelog/pull/1783, which is included Wavelog release 2.0.2.
Verified on DF instance with version 2.0.2 and same client details as mentioned above.
Thanks for the quick fix and release!

PR #1783 (2025-03-24): Fix iterator-iterating with safari (#1782)
Safari does not know (iterator).forEach(), so use a for-of loop here. Thanks @DC1TC for reaching out. Fixes #1782.
closed 2025-03-24 by DC1TC