wavelog/wavelog #1810: Option to add Custom Hams of Note

DA1EE, 2025-03-26:
**What Feature is Missing?**
Hi, I really like the implemented Ham of Note feature. I would really enjoy, if I could generate my own "Hams of Note" file (one or even a couple), and add it as a reference. Similar to the function of PoLo. This would allow me, to update my own files with hams of note.

**Why is This Feature Important?**
I would no longer be dependent on the HAM2K Hams of Note file, but use other files as well. E.g. from my local club, contest clubs or someone else.

**Is This Feature Personal or Beneficial to Others as Well?**
It would be beneficial for a lot of people.

int2001, 2025-03-27:
You can adjust the URL where the list is fetched only by altering the code here: https://github.com/wavelog/wavelog/blob/ebeaec4e3a4c2ffc6303098e9639b46e00aca9bb/application/models/Update_model.php#L490

Hope this helps at least for the first time, until we possibly(!) find another solution

DA1EE, 2025-03-27:
Hi, thanks for the hint.  I might add a copy of that file, and test that for myself, but on the other hand, I might just wait, until it is available, if it will come anytime.

I do understand, that it does not have highest priority.

int2001, 2025-03-28:
Feature is available in dev.

closed 2025-04-06 by github-actions[bot]