wavelog/wavelog #1833: QSO of the day box in the Edit Export Map options.

f4axk, 2025-04-01:
Hello team!
Would it be possible to add a QSO of the day box
in the Edit Export Map options.
Useful for displaying only the QSOs of the day to the visitor.

73 F4AXK laurent

phl0, 2025-05-15:
Are we talking abou the public dashboard view or the mapping feature?

f4axk, 2025-05-15:
Hi PHL0

The mapping function, if possible.
Or add a function to display the day's QSOs in
Widgets Static Map Images.

The mapping function only allows you to enter a number of QSOs
and in the Widgets Static Map Images (start_date and end_date).

The function is to display the day's QSOs either in the mapping
or in the Widgets Static Map Images.

Example: https://mon-wavelog/index.php/visitor/exportmap/F4AXK
or for the wigget
https://[wavelog url]/index.php/staticmap/render/[slug]?qsoday

I hope I'm precise enough.

Thanks 73s to the whole F4AXK team

f4axk, 2025-05-16:
> Ajout d'un paramètre appelé `day`à la carte statique qui permet `today`ou `yesterday`et ne montre que ces QSO.

Thanks for this new "day=today/yesterday" function.
I just modified and tested it; it works great.

73s f4axk

closed 2025-05-25 by github-actions[bot]