wavelog/wavelog #1843: Easiest Way to Install on Win 11

soakland5, 2025-04-04:
Hello!

I am a Cloudlog user (web version) and love it. I'd like to set up my own Win 11 server for Wavelog but not sure what the easiest way is.

Ideas?

Thanks
Scott, N1IA

AndreasK79, 2025-04-04:
Hi, 

Are you planning on running this locally on your home machine, or do you have an actual windows server you are planning to run this on?

soakland5, 2025-04-04:
Hi, Just locally on home machine.

AndreasK79, 2025-04-04:
I see. I personally am running Xampp to run and develop Wavelog on locally.
Another option is to run Wamp. Docker is also an option that is possible.

Some tips for Wamp is found here.

For Xampp, it is pretty much to install it, and place Wavelog in a folder under htdocs.

soakland5, 2025-04-04:
Thanks for the info. :)

AndreasK79, 2025-04-04:
No problem. Good luck, and let me know if you need some help.

closed 2025-04-07 by AndreasK79