wavelog/wavelog #1850: Same county in all caps vs mixed upper and lower case counted as different counties

Reid-n0rc, 2025-04-07:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
The US Counties display (Awards > USA > US Counties) counts the same county stored in all caps as a different county than the same county name stored in mixed case. 

**Version Information**
- Wavelog:         1.9.1

**Steps to Reproduce**
1. Go to 'Awards > USA > US Counties'
2. Click on any 'Counties Worked'
3. See that counties with the same name but different case are counted as separate counties.

[image]

**Expected Behavior**
Same county should be counted as the same, regardless of county name case.

phl0, 2025-04-08:
I am sorry. This is an issue with data obviously imported from another logging software. It is not a case of case sensitivity here but of spelling. See "AK,ALEUTIANS WEST" vs. "AK, Aleutians West". Here it is the space after the comma that is making the difference. Sorry this cannot be solved in software as we cannot cover all cases of spelling delivered by 3rd party software. 

The ADIF standard states that there is no space between the state and the county name: https://www.adif.org.uk/315/ADIF_315.htm#Sponsor_Defined_Code_Format. So I fear you need to fix the counties entered wrongly.

Reid-n0rc, 2025-04-08:
@phl0 Thanks for pointing that out. I'll fix my log.

closed 2025-04-08 by Reid-n0rc

AndreasK79, 2025-04-08:
@Reid-n0rc you also have one that says AK,AK,KENAI PENINSULA, in case you you didn't notice.