wavelog/wavelog #1851: When editing QSO, county dropdown not available

Reid-n0rc, 2025-04-07:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
When trying to edit a saved QSO and change the county information, the county drop down is not available. The only way I can get this to populate the county is to run "Update from Callbook" (in this case it pulls from QRZ) and save. After I go back into the QSO record, the county is now there.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu

- Wavelog:      1.9.1

**Steps to Reproduce**
1. Edit an existing QSO record that has had details populated by callbook data. 
2. Delete the Awards > US County information.
3. Save the record.
4. Edit the same record, go to the awards page. 
5. See that the drop down is not available for the US County field.
6. See that typing a county will not remain in the field.
7. Click 'Update from Callbook'
8. Click 'Save changes'
9. Edit the same record again and go to the awards page.
10. See that the county data is now populated.

 
**Expected Behavior**County**Screenshots**nclude screenshots if applicable to illustrate the issue.
[image]

**Desktop Information:**
- OS:               iOS Sequoia 15.3.2
- Browser:         Firefox ver 137.0

PopeFelix, 2025-05-09:
I can reproduce this in Wavelog v2.0.3. Looking into it.

PopeFelix, 2025-05-20:
Also present in v2.0.4

PopeFelix, 2025-05-21:
Fixed in #1978

closed 2025-05-27 by Reid-n0rc