wavelog/wavelog #1860: eQSL Cards unable to be viewed/downloaded when in eQSL Archive

bcshort, 2025-04-09:
When you have archived eQSL entries on the eQSL site, Wavelog is unable to download the eQSL card, giving the error "The requested content cannot be loaded, please try again later". Right clicking on the view/download button and opening in a new tab/window gives the message "Rate Limited"

**Version Information**
- Wavelog:          2.0.3 (Docker Container)

**Steps to Reproduce**

1. Account dropdown, 3rd party services,  eQSL Import/Export
2. Select the "Download eQSL cards Tab"
3. Press view/download button on a card that your know is in your eqsl.cc archive

**Expected Behavior**
Card to display/be downloaded

**Screenshots**
N/A - see initial description.

**Desktop Information:**
- OS:              MacOS
- Browser:          Safari, Firefox
- Version:          Safari 18.1, Firefox 137.0.1

**Additional Context**
Only appears to be when eQSL is in it's archive folder. eQSL cards that are in the eQSL inbox download fine.
Have verified that I can manually view card from archive.

**Workaround**
Manually downloading cards from eQSL and editing each Wavelog entry to upload QSL card :(

HB9HIL, 2025-04-10:
I'm not sure if we can provide a solution here. Isn't it possible to move all entries back to INBOX on eQSL itself?

HB9HIL, 2025-04-11:
I'm not 100% sure about this as I didn't write the eQSL stuff. May @phl0 has some more Information here. 

If you are familiar with SQL you could reset those fields in the database or you edit a QSO and set eQSL received to "No" which should trigger a recheck of the QSO in the next sync round.

bcshort, 2025-04-11:
Thanks Fabian,

I'm going ahead and resolving this based on the information you provided leading to another discovery that appears to ultimately be the issue.

It looks like the previous logging application (you could say it was deluxe...) exported the ADIF with the start time of the contact being in Local time, and end time being in UTC. Once I set the eqsl to received = "no" and did a check, it quickly became apparent what the issue was.

I have confirmed that once these details were correctly matching, the QSL card was then able to be downloaded.

Now I have the fun of going through 550 log entries and fixing the Time Start fields.

Thanks for your patience :)

closed 2025-04-11 by bcshort