wavelog/wavelog #1864: Add "Update Distance Data" to Cron Jobs [minor enhancement]

DA1EE, 2025-04-12:
**What Feature is Missing?**
Hi, it would be a nice feature, if you could add the "Update Distance Data" to the list of cronjobs. 

**Why is This Feature Important?**
I am doing a lot of POTA. And the distance data is not stored, when logging via Ham2K PoLo. But wavelog can calculate it. I have to be site admin and run this manually to update the distance data.

**Is This Feature Personal or Beneficial to Others as Well?**
I think others would benefit from this as well.

Thank you in advance, and thank you very much for all your effort, into making Wavelog such a great logging software.

73 de da1ee Lucas

int2001, 2025-04-12:
Simply go to analytics->distances and the missing distances will be updated. 

Use the admin -> update distances with extreme care. It updates every (!) QSO on the instance.

int2001, 2025-04-12:
To describe my own workflow when doing POTA with Polo or without WL:

1. Log at $app
2. import ADIF to WaveLog
3. press update from Callbook at Logbookadvanced (updates dist as well if empty b4 and grids are present)
4. finished

The adminfunction for updating ALL QSOs of a whole instance was nearly forgotten. Thanks for the heads-up. We are going to remove it in a future version. Too dangerous. Perhaps we're going to add an extra-button (besides update from Callbook and the analytics->distance thing) where a user can update only HIS distances while logged in.

At the moment i can't see a cron-feature for it, because cron is always instance-wide. Perhaps we find a fourth way somewhere in the future.

DA1EE, 2025-04-12:
@int2001 : Thank you. I understand your workflow, and I am working a similar one. Just used the instance wide all qso function, as I was not aware of the other function. Will adapt this.
Of course I was hoping for an automated solution, not to worry about anything, and reducing manual work. But I can follow your intention behind your decision.

When I asked for the feature, I expected a quite small effort (just run another command in a cycle), with the background info you provided, I can fully understand, that this feature does not have the priority, due to the effort, which would be needed (starting with a concept, how this could be actually solved).

Thank you very much and have a great weekend!