wavelog/wavelog #1867: When entering POST QSO data, updating the POTA location does not update the county location.

Reid-n0rc, 2025-04-12:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
When I am entering QSO data from the Post QSO screen county information is populated by a pull from QRZ.com after I enter a callsign. Next, I enter the POTA information. Location, Grid, and State information is all changed when the park information is pulled in, but the original county data pulled in from QRZ remains.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu

- Wavelog:          1.9.1

**Steps to Reproduce**
**Must have QRZ xml data enabled**

1. Go to 'QSO > Post QSO Data'
2. Enter 'KC0QLU' in the Callsign box. 
3. Click on 'General' tab.
4. Note that this station is in Clinton county MO.
6.  Enter 'US-2375' in the POTA box
7. Note that Location, Grid, and State all update with POTA location.
8. See error: County still says 'Clinton.'

**Expected Behavior**
County information should update with the other information.

**Desktop Information:**
- OS:               macOS 15.3.2 (24D81)
- Browser:          Firefox
- Version:          137.0.1

int2001, 2025-04-12:
indeed. the pota-api (where Wavelog fetches grid, location and state) does not provide the county in a direct way.
there's a key called `locationName` where "Louisana" is contained for US-2375.
See: https://api.pota.app/park/US-2375

i am not sure, since it is a undocumented API from the POTA-Guys, if there's always the correct county in it.

perhaps someone from the pota-guys can tell us more? do you know one?

Reid-n0rc, 2025-04-13:
They do provide lat lon, so a lookup with the FCC census API could provide the county data, at least until I can convince the POTA admins to add county data.

https://geo.fcc.gov/api/census/

int2001, 2025-04-14:
good find. technically this could work.
but, tbh: we are not keen to advertise things from the FCC within Wavelog as their Terms-of-use require. See here: https://www.fcc.gov/reports-research/developers/api-terms-service

for now i'd like to close this issue, since it isn't a bug. the information about the county isn't provided in a direct way by the pota api. so there's no direct way to implement it. as a woraround we could offer to empty the county-field automatically if you enter a POTA-Ref.