wavelog/wavelog #1874: Localize checkIfWorkedBefore in contest logging to reduce server load.

Beiri22, 2025-04-14:
**What Feature is Missing?**
I was trying to use the DARC wavelog instance and noticed that they disabled contest logging. This was due to the high server load as each keystroke will call the server. As the scp call is done with regular and contest logging; and checkIfWorkedBefore only when contest logging, I think they referred to checkIfWorkedBefore. The feature would be to use a local cache in order to determine checkIfWorkedBefore offline during contest.

** How? **
When beginning the contest session / opening the contest logging page, you pull all the logs of this contest from server (only needed fields: callsign, band, time). While the page is open and new qso's are logged, the local data storage (may simply be a HashMap variable) is also updated; and it can be used to checkIfWorkedBefore without asking the server each time.

**Why is This Feature Important?**
Reduce server load during contest on large instances.

**Is This Feature Personal or Beneficial to Others as Well?**
This would be beneficial to all users, because it would reduce server load; and hopefully convince DARC to activate contest logging feature.

HB9HIL, 2025-04-14:
The contest logging feature is disabled for other reasons aswell at DARC. I'm currently working on a refactoring of the contest stuff but you'll need some more patiente (probably summer/autuum 2025) as I have family and a job :-D haha

Beiri22, 2025-04-14:
They told me only that one. I really appreciate your work; and I know what family+job means to private projects. From other projects I know that a tracking issue could be helpful to give users a rough information on what is going to happen; what are open points of discussion, what parts of the implementation / rework are planned and what parts are already done. Maybe you would find enthusiastic users with programming skills that can help in discussion and implementation...

HB9HIL, 2025-04-14:
I got some ideas how things may work in near future. Please be patient a few more weeks :)

int2001, 2025-04-14:
Hi Beiri. There are more reasons I disabled that on the DARC instance than this small one.

Fact is: the current contestmodule at wavelog is not really a full contest module. Calculation of points - as an example - are missing. DARC wants to minimize support and give the users the best experience. As @HB9HIL already mentioned the Wavelog-Team is going to refactor the Contest-Section when there's time for it.

You can imagine that there will be tons of support-questions if DARC would enable that in the current status.

Now you've some background. If you want to use contesting within Wavelog you can still use a selfhosted instance.

Furthermore with your good ideas: Always welcome to do PR's against the dev and help to improve the contest-thing.

PS: If you have special-questions regarding the DARC-Instance (known as "Wavelog@DARC") please use logbuch@darc.de for questions.