wavelog/wavelog #1891: SAT azimuth & elevation keep getting updated when logging previous QSO manually

violarulan, 2025-04-18:
If we try to log a SAT QSO of which satellite has orbited beyond the horizon, the azimuth and elevation input fields cannot be set manually. They keep getting updated with current value, preventing to specify them. 

[image]

In https://github.com/wavelog/wavelog/commit/2dec053ddb93a9819feebb8972c15d90c2ba4ee0, we added a timer to update it automatically.

It would be nice to add a checkbox or something to enable it manually.

[image]

int2001, 2025-04-18:
You can use post-QSO. Enter the time of the QSO and the az/ele should be correct.

violarulan, 2025-04-18:
Thanks for the tip. That works perfectly.

closed 2025-04-18 by violarulan