wavelog/wavelog #1895: Request: Auto-fill Band and Mode from previous QSOs when adding contacts

krisp, 2025-04-22:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
The Add QSO page automatically looks up grid and name and location and other pertinent details when you type in a remote station's callsign into the form. It would be nice if it also automatically filled in the band and mode from the last contact with the station. 

**Why is This Feature Important?**
I typically work the same group of people on the same bands. The first QSO of the morning always comes up at 160 meter and AM mode because they are at the top of the list. Subsequent QSOs added from the same form keep this filled in, but it would be nice if after typing the callsign when it looks up the grid and name and such, if it also automatically filled in the most recent band and mode for the contact. 

**Is This Feature Personal or Beneficial to Others as Well?**
It seems like this could be useful to anyone who works the same folks on a daily basis. The current behavior is fine after the first contact is logged (band and mode sticking for new contacts) but having it update for the first contact when the defaults are selected would be a nice-to-have.

HB9HIL, 2025-04-22:
I really don't see how this is usefull. If you are currently on 20m/ssb and you type in a callsign you hear on 20m/ssb, which was worked by you last on 15m/ssb, it would be really cumbersome if Wavelog would switch band/mode based on that....

I really can't get why this is nice-to-have.. sorry..

HB9HIL, 2025-04-22:
For a quick switch between different band/mode I'd recommend to use "Favourites" in the QSO logging window

[image]

krisp, 2025-04-25:
I didn't know about the favorites, I'll give that a shot. It's really only useful at first lookup time where it defaults to AM/160 meters. Once QSOs have been logged it makes sense for it to stick to the same band. Perhaps a user setting to set a "default" band and mode for whenever it resets back to the first in the list would be good.

HB9HIL, 2025-04-25:
I once thought about that. But in fact we recommend to use Favs and CAT Control together with WLgate. Then you don't need a default value here :)