wavelog/wavelog #1909: Automated upload of SOTA logs to sotadata.org.uk

corvus-ch, 2025-04-27:
**What Feature is Missing?**
Automated upload of SOTA logs to [sotadata.org.uk](https://www.sotadata.org.uk).

**Why is This Feature Important?**
Wavelog has spoiled me with the fact that logs get automatically uploaded to several services: LoTW, ClubLog, eQSL and QRZ. sotadata.org.uk is missing on that list.

Because of that, my current workflow requires me to manually export the CSV and manually upload it to on sotadata.org.uk.

**Is This Feature Personal or Beneficial to Others as Well?**

This feature is interesting for all Wavelog users also doing SOTA.

PopeFelix, 2025-05-09:
I don't see an upload API for SOTA. I've reached out to their IT team to see if such a thing exists, but if it doesn't, I don't think this is going to be possible.

PopeFelix, 2025-05-16:
I've confirmed that there is a SOTA API, so I'll start working on this once I've finished on #1851

winnieXY, 2025-11-17:
Any news on that? That would be a feature which I really like to have within wavelog :-)