wavelog/wavelog #1911: ER: Add user specific option to disable "no QSOs today" message.

DC1TC, 2025-04-27:
**What Feature is Missing?**
A switch to disable the "no QSOs today" message:
> You have made no QSOs today; time to turn on the radio!

Ideally it would be a personal preference and not server-wide.

**Why is This Feature Important?**
Seems like people feel patronized by this "call to action".

**Is This Feature Personal or Beneficial to Others as Well?**
At least three people in this thread, also some comments in the Draussenfunker Discord.
https://troet.cafe/@warteschleife/114408905427912535

DC1TC, 2025-04-27:
Looks like there is a setting, but it's global?

https://github.com/wavelog/wavelog/blob/ebed76cbf1d71b0d719021eb04501b73f1d79eba/application/views/options/appearance.php#L60-L66

HB9HIL, 2025-04-27:
Hello

yes it's global and can be set in Admin > Appearance > Dashboard Banner

AndreasK79, 2025-04-28:
#1913

AndreasK79, 2025-04-28:
The code is now in dev.