wavelog/wavelog #1928: QSL Log 500 Error

jwussler, 2025-05-05:
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Ubuntu 24.04.2 LTS
- PHP:              8.3.6
- MySQL         8.0.41-0ubuntu0.24.04.1
- Wavelog:      2.0.3
I am on the DEV branch 
Anytime I go to the QSL que tab the sever tosses a 500 error i can I can duplicate the same error on a fresh install of wave log.

here is the output of the error log 

<?php defined('BASEPATH') OR exit('No direct script access allowed'); ?>

ERROR - 2025-05-05 05:52:54 --> Severity: error --> Exception: Expression #219 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'wavelog.o.id' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by /var/www/html/system/database/drivers/mysqli/mysqli_driver.php 301

phl0, 2025-05-05:
Fix for this is in dev branch (see https://github.com/wavelog/wavelog/pull/1929).

jwussler, 2025-05-05:
Thanks!

Works as expected now

PR #1929 (2025-05-05): Fix SQL group by error in QSL queue
closes #1928
closed 2025-05-05 by jwussler