wavelog/wavelog #1932: lower case table names for Window and Linux datebase compatibility

uiolee, 2025-05-05:
## **What Feature is Missing?**

**use lower case table names all the time**

for compatibility between Windows (case-insensitive) and Linux (case-sensitive) systems.

## **Why is This Feature Important?**

Using consistent lowercase table names avoids issues during database exports/imports, particularly when migrating from or to systems with different `lower_case_table_names` settings in MySQL/MariaDB. This change can prevent bugs and unexpected behavior due to case mismatches in SQL queries.

> refs: 
>
> https://dev.mysql.com/doc/refman/8.4/en/identifier-case-sensitivity.html
> https://dev.mysql.com/doc/refman/8.4/en/server-system-variables.html#sysvar_lower_case_table_names

`lower_case_table_names` On Linux and other Unix-like systems, the default is 0. 
`lower_case_table_names` On Windows the default value is 1.

It means in default setting, 
- When we store tables in (or migrate to ) Window's database, the table name will be transformed to lower case and be stored. AND the SQL will ignore case. SO, no error will occur.

- BUT if we need migrated database from Window to Linux, The table name is still in lower case. BUT the SQL is case-sensitive. SO, it could not found the expected table.

In the meantime, the table name case change is irreversible. You can only delete the database and then change the settings before rebuilding the database.（OR, Manually modify table names in the database）

IMHO, consistent lowercase is a simple way to resolve this problem.
(I'm sorry, I'm not sure if the capitalized table name is used in wavelog for some necessary reasons)

## **Is This Feature Personal or Beneficial to Others as Well?**

I think this is beneficial to all users working in cross-platform environments or deploying to Linux servers after local development on Windows.

uiolee, 2025-05-05:
When I migrated my wavelog instance from Window to Linux, Wavelog always failed to run.

It took me some time to investigate before realizing that this is a problem with the database. This is related to `lower_case_table_names` of the database.

[image]

----------
----------

[image]

[image]

In the end, my solution was to rework the table names that had been converted to lowercase to their original names, and wavelog finally got running!

PopeFelix, 2025-05-09:
This seems straightforward enough. Let me take a crack at it.

PopeFelix, 2025-05-09:
Can you give me a little more detail of how you're exporting your logs? I don't see any functionality (in the demo app, at least) for executing (e.g.) `mysqldump`

PopeFelix, 2025-05-09:
> This seems straightforward enough. Let me take a crack at it.

No, I am a fool. This is a massive pain, and I'm not sure it's a supported use case. Until I get confirmation that it *is* a supported use case, I'm not going to mess with it.

uiolee, 2025-05-10:
> Can you give me a little more detail of how you're exporting your logs? I don't see any functionality (in the demo app, at least) for executing (e.g.) `mysqldump`

https://github.com/wavelog/wavelog/blob/ebed76cbf1d71b0d719021eb04501b73f1d79eba/application/config/config.sample.php#L300

PopeFelix, 2025-05-12:
I'm sorry, I don't follow you. What is it, please, that you're trying to tell me?