wavelog/wavelog #1933: When operating on the QSL information of QSO in "/search", the front end shows an incorrect response

vastsea-wuji, 2025-05-05:
When operating on the QSL information of QSO in "/search", the front end shows an incorrect response. And the operation did not take effect

**Describe the bug**
I use wavelog to manage my QSL. After I receive a qsl card, I search for this qso in the search box on the home page and try to change its qsl status. However, when I click "Mark Received qsl (Direct Mail)", the qsl receipt mark will turn green, which is normal. But the qrz sending status will change to yellow. And after re-searching this qso, you will find that none of these states have been changed

**Version Information**

- OS:              centos nginx/1.24.0
- PHP:              8.1.31
- MySQL:   5.7.44-log
- Wavelog:          2.0.2 

**Steps to Reproduce**
1. Go to index
2. Search for the call sign in the search box at the top

[image]

3. Click the "Mark Received Card (Direct Mail)" button corresponding to QSO

[image]

4. See error

[image]

5. Refresh the page, but the status received by QSL remains unchanged

[image]

int2001, 2025-05-06:
Known bug in 2.0.2. Pse upgrade to 2.0.3

vastsea-wuji, 2025-05-06:
Thank you for your reply. Since I installed wavelog by uploading the source code, I didn't upgrade in time. I'm very sorry for providing the bug in the old version. I will upgrade as soon as possible to solve it

closed 2025-05-06 by vastsea-wuji