wavelog/wavelog #1944: FT4 QSO Mode set to MFSK in CSV export of QSL printing queue

flothom, 2025-05-11:
When exporting QSOs in the QSL printing queue to CSV, using the URL ...https://wavelog.<host>/index.php/qslprint/exportcsv/All, in the resulting CSV file, FT4 QSOs are set to mode MFSK instead of FT4.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- Server-Software | Apache
- PHP-Version: 8.2.28 OK
- MySQL-Version: 10.6.19-MariaDB
- Codeigniter Version: 3.2.0-dev
- Wavelog:  2.03

**Steps to Reproduce**
This only can be reproduced if there are FT4 QSOs in the QSL Queue.
1. Go to QSL Queue
2. Click on "Export requested QSLs to CSV-file"
3. Check the mode in the FT4 QSOs in the downloaded csv file

**Expected Behavior**
For FT8 QSOs, the mode is set to FT8 which is correct. Therefore, I expect that for FT4 qsos, the mode in the CSV file is set to FT4 instead of MFSK. 

**Output Example (callsign removed)**
`<my callsign>,<DX callsign>,,"2025-01-01 10:10:10",MFSK,14080981,20M,+15,,,"PSE QSL",,<callsign>,27,BELARUS
`

HB9HIL, 2025-05-11:
Hello @flothom 

This is correct. FT4 is no "MODE" in Adif Standard, rather it is a "SUBMODE" of MFSK. In the CSV export there is the main mode exported why it shows MFSK

Definition see here: https://adif.org/315/ADIF_315.htm#Mode_Enumeration

HB9HIL, 2025-05-11:
PS: Even FT8 is technically also MFSK in Adif Standard it's enumarated as MODE instead. Dunno why :-D

flothom, 2025-05-11:
OK, then, thanks for explanation, @HB9HIL  - I was not aware of that.
Closing this issue.
Thanks, Florian

closed 2025-05-11 by flothom

HB9HIL, 2025-05-11:
@flothom However it doesn't hurt so I added the submode in the CSV export :) will be in the next release