wavelog/wavelog #1980: Post QSO - Band field cleared when frequency is changed only on 4m band

redgeler, 2025-05-21:
Hello, When trying to post some QSOs on the 4m Band. I've hit an issue where upon entering the frequency it cleared the band field, preventing the QSO from being saved.

As a work-around, I accepted the default frequency of "70.2" and then edited the QSO to update the frequency.

Also found the same behaviour for the 1.25cm Band.

I have tested this with two separate servers/instances of Wavelog. 
Tested using two desktop computers with multiple browsers. Edge, Firefox and Chrome.

**Version Information**
Instance1
- OS: Ubuntu 24.04.2
- PHP: 8.3.6
- MariaDB: 10.11.11-MariaDB-0ubuntu0.24.04.2
- Wavelog: 2.0.3

Instance2
- OS: Ubuntu 24.04.2
- PHP: 8.3.6
- MariaDB: 10.11.11-MariaDB-0ubuntu0.24.04.2
- Wavelog: 2.0.4

**Steps to Reproduce**
1. Go to "Post QSO"
2. Select the "4m" Band (The Frequency is automatically set to "70.2")
3. Change the Frequency  e.g. "70.3"
4. Do anything to change the focus away from the the Frequency field
5. The Band field is cleared.

**Expected Behaviour**
The Band to remain populated in the Band field.

**Screenshots**
Select the 4m Band
[image]
Change the Frequency
[image]
When focus is lost the band is cleared.
[image]

**Desktop Information:**
- OS: Windows 10
- Browser:  Edge, Chrome, Firefox
- Version:   136.0.3240.76, 136.0.7103.114, 138.0.4

Is there any further info I can provide?

Rob.

phl0, 2025-05-21:
Fix proposed in https://github.com/wavelog/wavelog/pull/1982.

phl0, 2025-06-03:
I have widened the 4m band to 75MHz with https://github.com/wavelog/wavelog/commit/c970b07bf7e7c40beed0807848880e9c7e66c63b.