wavelog/wavelog #1988: Cannot install wavelog - database migration error

tuxmartin, 2025-05-23:
Hi,

I’m trying to install Wavelog using Docker, but it keeps failing on the database migration.

I’m using the latest Docker image (pulled on May 22, 2025).

My database is external to Docker: MariaDB 10.11.11-MariaDB-0+deb12u1-log.
The database connection test succeeds and the installer creates the tables, but then it fails during migration.

What am I doing wrong?

**Version Information**
- OS:  Debian 12
- MySQL/MariaDB: 10.11.11-MariaDB-0+deb12u1-log

Docker image pulled 22.5.2025:
```
REPOSITORY                TAG       IMAGE ID       CREATED      SIZE
ghcr.io/wavelog/wavelog   latest    21c74391baeb   6 days ago   720MB
```

```
# docker-compose.yml 
services:
  wavelog-main:
    container_name: wavelog-main
    image: ghcr.io/wavelog/wavelog:latest
    environment:
      CI_ENV: docker
    volumes:
      - /opt/wavelog/config:/var/www/html/application/config/docker
      - /opt/wavelog/uploads:/var/www/html/uploads
      - /opt/wavelog/userdata:/var/www/html/userdata
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro
    ports:
      - "80:80"
```

**Screenshots**

[image]

[image]

[image]

[image]

tuxmartin, 2025-05-23:
I forgot to post log:

```
                            Wavelog Installer Debug Log
-------
Log Location: ../application/logs/installer_debug.log

INFO - 2025-05-22 19:06:35 --> Called Installer index.php
INFO - 2025-05-22 19:06:35 --> With URL: http://wavelog.example.net/install//
INFO - 2025-05-22 19:06:35 --> From IP: 10.123.1.66
INFO - 2025-05-22 19:06:35 --> Set language cookie to "Czech"
INFO - 2025-05-22 19:06:47 --> Called Installer index.php
INFO - 2025-05-22 19:06:47 --> With URL: http://wavelog.example.net/install//
INFO - 2025-05-22 19:06:47 --> From IP: 10.123.1.66
INFO - 2025-05-22 19:06:47 --> Set language cookie to "English"
ERROR - 2025-05-22 19:08:49 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:08:51 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:08 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:09 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:09 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:09 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:09 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:20 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:21 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:29 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:09:29 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:11:10 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:11:21 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:11:21 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:11:21 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:11:21 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:11:21 --> Database Check Error: Connection refused
ERROR - 2025-05-22 19:11:22 --> Database Check Error: Connection refused
DEBUG - 2025-05-22 19:13:40 --> Start writing config.php
INFO - 2025-05-22 19:13:41 --> Data validation passed.
INFO - 2025-05-22 19:13:41 --> CI_ENV is set to docker. Using docker config.php config path.
INFO - 2025-05-22 19:13:41 --> config.php template file read successfully.
INFO - 2025-05-22 19:13:41 --> Config.php file prepared successfully. Writing to file...
INFO - 2025-05-22 19:13:41 --> config.php file written successfully.
DEBUG - 2025-05-22 19:13:41 --> File: config.php successfully written
DEBUG - 2025-05-22 19:13:41 --> Start writing database.php
INFO - 2025-05-22 19:13:42 --> Data validation passed.
INFO - 2025-05-22 19:13:42 --> CI_ENV is set to docker. Using docker database.php config path.
INFO - 2025-05-22 19:13:42 --> database.php template file read successfully.
INFO - 2025-05-22 19:13:42 --> Database config file prepared successfully. Writing to file...
INFO - 2025-05-22 19:13:42 --> database.php file written successfully.
DEBUG - 2025-05-22 19:13:42 --> File: database.php successfully written
DEBUG - 2025-05-22 19:13:42 --> Start creating database structure with assets/install.sql
INFO - 2025-05-22 19:13:42 --> Data validation passed.
INFO - 2025-05-22 19:13:42 --> SQL queries prepared successfully. Writing to database...
INFO - 2025-05-22 19:13:42 --> Database tables created successfully.
DEBUG - 2025-05-22 19:13:42 --> Tables successfully created
DEBUG - 2025-05-22 19:13:42 --> Start migrating database to the newest version.
ERROR - 2025-05-22 19:13:42 --> Could not migrate database. Ajax crashed. Status: 0 Status Text: error
DEBUG - 2025-05-22 19:14:03 --> Start writing config.php
INFO - 2025-05-22 19:14:04 --> Data validation passed.
INFO - 2025-05-22 19:14:04 --> CI_ENV is set to docker. Using docker config.php config path.
INFO - 2025-05-22 19:14:04 --> config.php template file read successfully.
INFO - 2025-05-22 19:14:04 --> Config.php file prepared successfully. Writing to file...
INFO - 2025-05-22 19:14:04 --> config.php file written successfully.
DEBUG - 2025-05-22 19:14:04 --> File: config.php successfully written
DEBUG - 2025-05-22 19:14:04 --> Start writing database.php
INFO - 2025-05-22 19:14:05 --> Data validation passed.
INFO - 2025-05-22 19:14:05 --> CI_ENV is set to docker. Using docker database.php config path.
INFO - 2025-05-22 19:14:05 --> database.php template file read successfully.
INFO - 2025-05-22 19:14:05 --> Database config file prepared successfully. Writing to file...
INFO - 2025-05-22 19:14:05 --> database.php file written successfully.
DEBUG - 2025-05-22 19:14:05 --> File: database.php successfully written
DEBUG - 2025-05-22 19:14:05 --> Start creating database structure with assets/install.sql
INFO - 2025-05-22 19:14:05 --> Data validation passed.
[... 112 lines cut ...]
DEBUG - 2025-05-22 20:27:47 --> Tables successfully created
DEBUG - 2025-05-22 20:27:47 --> Start migrating database to the newest version.
ERROR - 2025-05-22 20:27:47 --> Could not migrate database. Ajax crashed. Status: 0 Status Text: error
DEBUG - 2025-05-23 08:08:51 --> Start writing config.php
INFO - 2025-05-23 08:08:52 --> Data validation passed.
INFO - 2025-05-23 08:08:52 --> CI_ENV is set to docker. Using docker config.php config path.
INFO - 2025-05-23 08:08:52 --> config.php template file read successfully.
INFO - 2025-05-23 08:08:52 --> Config.php file prepared successfully. Writing to file...
INFO - 2025-05-23 08:08:52 --> config.php file written successfully.
DEBUG - 2025-05-23 08:08:52 --> File: config.php successfully written
[... cut ...]
DEBUG - 2025-05-23 08:08:53 --> Start creating database structure with assets/install.sql
INFO - 2025-05-23 08:08:53 --> Data validation passed.
INFO - 2025-05-23 08:08:53 --> SQL queries prepared successfully. Writing to database...
INFO - 2025-05-23 08:08:54 --> Database tables created successfully.
DEBUG - 2025-05-23 08:08:54 --> Tables successfully created
DEBUG - 2025-05-23 08:08:54 --> Start migrating database to the newest version.
ERROR - 2025-05-23 08:08:54 --> Could not migrate database. Ajax crashed. Status: 0 Status Text: error
 ```

tuxmartin, 2025-05-23:
I tried official docker compose. I had the same problem.

https://github.com/wavelog/wavelog/wiki/Installation-via-Docker#docker-composeyaml

HB9HIL, 2025-05-23:
The current version 2.0.4 works with out any issues with our docker compose file

[image]

Please consider a clean start also with removing the already created volumes.

tuxmartin, 2025-05-23:
I removed all volumes, dropped database, created new database and granted all privileges. Still the same problem:

[image]

but I see tables in database, so connection is good. 

Latest wavelog 2.0.4 (`latest` docker tag).

HB9HIL, 2025-05-23:
The database migration runs directly via web-url calls

probably the final url is not reachable from your browser. Did you remove some lines in the installer debug log? Usually you can see from which location you access the installer. See the example in my installation a few moments ago

[image]

HB9HIL, 2025-05-23:
I think the issue is definitly on your side.. Do you use a reverse proxy?

tuxmartin, 2025-05-23:
Thank you!

Problem was at nginx reverse proxy configuration.

closed 2025-05-23 by tuxmartin

phl0, 2025-05-23:
> Problem was at nginx reverse proxy configuration.

Would be nice to leave a hint here for others who might run into the same problem ;-)

tuxmartin, 2025-05-23:
Probably the problem was `X-Forwarded-Proto` header. I had set `http` for `https` vhost.