wavelog/wavelog #1989: QSL queue filter, exclude electronic QSLs when printing labels

DL2CC, 2025-05-23:
**What Feature is Missing?**

When answering incoming bureau or direct cards I mark the QSL as requested bureau or requested direct.
When then going to the qsl queue to prepare printing QSL labels (or going to the label printing directly) the "requested electronic" qsos are included.

**Why is This Feature Important?**
Doesn't make sense to print QSL labels for electronic QSLs

**Is This Feature Personal or Beneficial to Others as Well?**
For all who still use paper QSLs.

DL2CC, 2025-05-23:
Just to add: it also seems that QSOs show as "QSL requested" for electronic QSLs when those QSLs have already been sent to via electronic QSL.

int2001, 2025-05-24:
Where (and why?) do you set "Requested Electronic"?