wavelog/wavelog #1994: "missing or wrong api key" when copied from API generation.

MikeWills, 2025-05-26:
I am trying to set up wavelog gate. When I do a test, I get "missing or wrong api key". So in troubleshooting, I tried doing a curl test based on the API documentation. Replacing "<MY KEY HERE>" with my actual key copy-pasted from the API settings in Wavelog. (see below). I get "{"status":"failed","reason":"missing or wrong api key"}"

```
Mikes-MacBook-Pro ~ % curl -X POST https://mydomain.com/index.php/api/qso -H "Content-Type: application/json" -H "Accept: application/json" -d "{\"key\":\"<MY KEY HERE>\",\"station_profile_id\":\"1\",\"type\":\"adif\",\"string\":\"<call:5>N9EAT<band:4>70cm<mode:3>SSB<freq:10>432.166976<qso_date:8>20190616<time_on:6>170600<time_off:6>170600<rst_rcvd:2>59<rst_sent:2>55<qsl_rcvd:1>N<qsl_sent:1>N<country:24>United States Of America<gridsquare:4>EN42<sat_mode:3>U/V<sat_name:4>AO-7<prop_mode:3>SAT<name:5>Marty<eor>\"}"

{"status":"failed","reason":"missing or wrong api key"}%
```

Any ideas on what I'm doing wrong? I also tried doing a capture in Wireshark and I'm not seeing any difference.

int2001, 2025-05-26:
i tested exactly the string from your post (but with my own key). Could not reproduce the behaviour you described.
Evven with a wrong station_id, i got another result (`{"status":"failed","reason":"station id does not belong to the API key owner."}`).

Please doublecheck the key you provide for 
- Spaces (there are no spaces in it)
- additional chars (accidently)
- length

you can also create a new key (and remove it after testing). please show us the new key and your curl-command with that key.

phl0, 2025-05-26:
> Here is the full curl command.
> 
Copied the command on my (Ubuntu) machine an it works:

```
{"status":"created","type":"adif","string":"","adif_count":0,"adif_errors":0,"messages":["Dryrun works"]}
```

Something fishy with the machine you are trying to run the curl from?

int2001, 2025-05-26:
ok. this is really strange.
tbh: everything looks okay. one thought was: perhaps you disabled the index.php at the URL (possible via config.php), but if that would be the reason, you won't even get a `{"status":"failed","reason":"missing or wrong api key"}` from the api.

which version of Wavelog you're using?
and what are you using as reverse-proxy?

phl0, 2025-05-26:
Hhmm. Not sure if DNS vould cause a wrong api key. But please delete the QSO I inserted into your database :-)

MikeWills, 2025-05-26:
What's the callsign? I don't see any for today.

phl0, 2025-05-26:
> What's the callsign? I don't see any for today.

N9EAT: 2019-06-16 1706z