wavelog/wavelog #2042: Not able to backup or export to adif

oliverbross, 2025-06-19:
Hi,

I have just tried exporting the log to Adif via the export function and it completely crashed my server.

Also, the same thing will happen if you use the backup function, it will crash services. So currently I'm not able to export/backup the log. 

Everything else works beautifully. My server details:

[image]

[image]

Wavelog is running smoothly, but the export/backup is not working.

What might be causing this?

Oliver

oliverbross, 2025-06-19:
I think those entries are related:

[Thu Jun 19 14:15:30.216351 2025] [proxy_fcgi:error] [pid 1603310:tid 136721531590336] (104)Connection reset by peer: [remote 188.121.161.30:50155] AH01075: Error dispatching request to :
[Thu Jun 19 14:15:30.216665 2025] [proxy:error] [pid 1603310:tid 136721531590336] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/17390881932456375.sock (*:80) failed
[Thu Jun 19 14:15:30.216672 2025] [proxy_fcgi:error] [pid 1603310:tid 136721531590336] [remote 188.121.161.30:50155] AH01079: failed to make connection to backend: httpd-UDS, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220187 2025] [proxy_fcgi:error] [pid 1603310:tid 136721615681216] [remote 188.121.161.30:50155] AH01067: Failed to read FastCGI header, referer: https://log.om0rx.com/index.php/adif
[Thu Jun 19 14:15:30.220203 2025] [proxy_fcgi:error] [pid 1603310:tid 136721615681216] (104)Connection reset by peer: [remote 188.121.161.30:50155] AH01075: Error dispatching request to : , referer: https://log.om0rx.com/index.php/adif
[Thu Jun 19 14:15:30.220244 2025] [proxy_fcgi:error] [pid 1603310:tid 136721548408512] [remote 188.121.161.30:50155] AH01067: Failed to read FastCGI header, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220251 2025] [proxy_fcgi:error] [pid 1603310:tid 136721548408512] (104)Connection reset by peer: [remote 188.121.161.30:50155] AH01075: Error dispatching request to : , referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220448 2025] [proxy:error] [pid 1603310:tid 136721531590336] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/17390881932456375.sock (*:80) failed
[Thu Jun 19 14:15:30.220454 2025] [proxy_fcgi:error] [pid 1603310:tid 136721531590336] [remote 188.121.161.30:50155] AH01079: failed to make connection to backend: httpd-UDS, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220661 2025] [proxy:error] [pid 1603310:tid 136721531590336] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/17390881932456375.sock (*:80) failed
[Thu Jun 19 14:15:30.220667 2025] [proxy_fcgi:error] [pid 1603310:tid 136721531590336] [remote 188.121.161.30:50155] AH01079: failed to make connection to backend: httpd-UDS, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220834 2025] [proxy:error] [pid 1603310:tid 136721531590336] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/17390881932456375.sock (*:80) failed
[Thu Jun 19 14:15:30.220839 2025] [proxy_fcgi:error] [pid 1603310:tid 136721531590336] [remote 188.121.161.30:50155] AH01079: failed to make connection to backend: httpd-UDS, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220872 2025] [proxy_fcgi:error] [pid 1603310:tid 136721640908480] [remote 188.121.161.30:50155] AH01067: Failed to read FastCGI header, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220880 2025] [proxy_fcgi:error] [pid 1603310:tid 136721640908480] (104)Connection reset by peer: [remote 188.121.161.30:50155] AH01075: Error dispatching request to : , referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220905 2025] [proxy_fcgi:error] [pid 1603310:tid 136721607272128] [remote 188.121.161.30:50155] AH01067: Failed to read FastCGI header, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.220912 2025] [proxy_fcgi:error] [pid 1603310:tid 136721607272128] (104)Connection reset by peer: [remote 188.121.161.30:50155] AH01075: Error dispatching request to : , referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.222549 2025] [proxy_fcgi:error] [pid 1603310:tid 136721674487488] [remote 188.121.161.30:50155] AH01067: Failed to read FastCGI header, referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.222560 2025] [proxy_fcgi:error] [pid 1603310:tid 136721674487488] (104)Connection reset by peer: [remote 188.121.161.30:50155] AH01075: Error dispatching request to : , referer: https://log.om0rx.com/index.php/qso?manual=0
[Thu Jun 19 14:15:30.226605 2025] [proxy:error] [pid 1603252:tid 136721523181248] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/17390881932456375.sock (*:80) failed

HB9HIL, 2025-06-19:
Are you running a plain install or a dockerized Wavelog?

oliverbross, 2025-06-19:
also, found this:

  GNU nano 7.2                                                                                                                                  php_log                                                                                                                                           
[19-Jun-2025 14:29:26 UTC] PHP Fatal error:  Allowed memory size of 536870912 bytes exhausted (tried to allocate 23085056 bytes) in /home/om0rx/domains/log.om0rx.com/public_html/application/views/backup/exportall.php on line 14

I'm running plain install ... looking at the memory in php now ... you gave me good suggestion, it's strange taht allowed max memory allocation is 512M and it tried to allocate 23MB and crasshed?

oliverbross, 2025-06-19:
no, no modifications, just plain install:

[image]

oliverbross, 2025-06-19:
I confirm this is still an issue, it worked once, when I try to use it gain, it crashed server. 

I can see partial file on my server, but missing qso data so not everything was exported:

31253190 Jun 19 14:53 logbook_2025_06_19_14_53_49.adi

So there is still an issue, maybe too many contacts in the log?

Many thanks.

int2001, 2025-06-21:
tested this with more than 50k QSOs on an instance, where mem is restricted to 256MB. No problems.
Are you using other software (e.g. Wordpress, etc.) on the same server?

perhaps some other PHP-Software claims all the memory.