wavelog/wavelog #2078: Discrepency between home page DXCC worked and Awards > International > DXCC

zcrowman, 2025-06-29:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
This morning I worked a new DXCC. On the home page DXCC total, the number jumped 2, but the number on the DXCC awards page incremented by 1 which I would have expected.
**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Apache
- PHP:              8.3.14
- MySQL/MariaDB:    8.0.28-0ubuntu0.20.04.3
- Wavelog:          2.0.5

**Steps to Reproduce**
1. Not sure how to reproduce because getting a new DXCC will take me a while.  But I know that in my case, 222 is the correct number, but the home page summary is showing 223 and it jumped from 221 before the new DXCC earlier this morning

**Expected Behavior**
No discrepency between the DXCC awards page and the home page DXCC summary

**Screenshots**
Added below

[image]
[image]

**Desktop Information:**
- OS:               MacOS Sonoma
- Browser:          Firefox
- Version:          140.0.1

**Smartphone Information:**
- Device:           [e.g., iPhone6]
- OS:               [e.g., iOS8.1]
- Browser:          [e.g., stock browser, safari]
- Version:          [e.g., 22]

**Additional Context**
Provide any other relevant context about the problem.

AndreasK79, 2025-07-02:
Seems the issue is related to dxcc = 0, which is used for /mm and /am. A fix is in the works by @phl0.

phl0, 2025-07-02:
Could you please check with latest dev if that solves your issues?

zcrowman, 2025-07-02:
>   phl0  left a comment   (wavelog/wavelog#2078) (https://github.com/wavelog/wavelog/issues/2078#issuecomment-3026775636)

Will do. Unfortunately I am travelling right now so I will need to do it when back.   
  
  
  
  

 Zachariah

AndreasK79, 2025-09-01:
@zcrowman are the numbers between dashboard and the DXCC award the same if you include the deleted ones in the DXCC award?

zcrowman, 2025-09-02:
> AndreasK79  left a comment   (wavelog/wavelog#2078) (https://github.com/wavelog/wavelog/issues/2078#issuecomment-3243748143)
>
>
> @zcrowman (https://github.com/zcrowman)  I found one thing that might do it. If by chance a QSO has the col_country set to invalid, it is excluded in the dashboard.
>
> You could open the logbook_model.php file in the application/models folder and remove line 3621 ($this->db->where('COL_COUNTRY !=', 'Invalid');) to see if the dashboard now shows the same as the DXCC award.

Hi Andreas. When I open that file, line 3621 does not contain the code you are showing. The file itself does contain 5 other instances of that code. I commented out all of those instances but the values on the dashboard didn't change. Still showing one too many
  
​
  
​Thanks for your efforts on this!
  
​
  
  
  
  
  
  
​_____________________________________
  
​Zachariah Crow
  
​VE7ZCZ

zcrowman, 2025-09-02:
Brilliant!! I can confirm that the dashboard now matches the DXCC award info. Thanks so much for taking the time to troubleshoot this rather unique problem.  Your efforts are very much appreciated. You have an amazing product there that I really enjoy using.

closed 2025-09-08 by zcrowman