wavelog/wavelog #2120: Req: 2-Factor-Authentication

Rakete2510, 2025-07-08:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
A 2-Factor-Authentication Feature is missing  like a TOTP (Timed-Based-One-Time Password)

**Why is This Feature Important?**
As an additional security level

**Is This Feature Personal or Beneficial to Others as Well?**
Its Beneficial for other aswell.

int2001, 2025-09-26:
jumping into this discussion.
Adding 2FA wouldn't mean "only add 2FA".
it means, that the API-Access has to be completly refactored. As long as one is able to modify/upload/read QSOs and things via API the 2FA would be senseless.

e.g.: Clublog has 2FA. The 2FA is absolutely nonsense there, because you're able to bypass it via the API. 
And tbh: We're not friends of snakeoil-security

int2001, 2025-12-31:
Honest Question: Why are people keen to complicate logins?
i mean: Keypass for easier(!) login would be interesting. But 2FA for a HamApplication?

However - perhaps in 2026