wavelog/wavelog #2125: Support enabling https in docker image via environment variables

blhoward2, 2025-07-09:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
Currently there does not appear to be a way to define a certificate file, enable https, and disable http via environment files in a docker compose file. While you can manually edit the .htaccess this would likely be lost on a rebuild of the image. Doing so through environment variables is a common method in docker. 

**Why is This Feature Important?**
Https is important for security and already recommended in the wiki. But for docker users there doesn't appear to be an easy way to enable it. 

**Is This Feature Personal or Beneficial to Others as Well?**
The benefits of https, and enabling people to enable it easily, are hopefully self-evident.

blhoward2, 2025-07-09:
Yes, except that I don't need a reverse proxy if I don't intend to expose it to the internet, but I still want the login token to last for 30 days.

HB9HIL, 2025-07-09:
A reverse proxy doesn't mean you have to expose it to the internet.. You can still use it locally

run your docker container - wavelog will be exposed per default on 8086

the install nginx and use the reverse proxy configuration from our wiki [here](https://github.com/wavelog/wavelog/wiki/Installation-via-Docker#reverse-proxy)

No SSH access? No problem! You also can extend your docker compose file with a reverse proxy like nginx or traefik. There are a bunch of tutorials in the web how this works. 

Don't get me wrong. I fully agree that HTTPS is mandatory. But we are here on the application layer. We provide the software. While our ressources are limited we can't take care of secure webhosting aswell. This is in the responsibilty of the user.

HB9HIL, 2025-07-09:
@blhoward2 No... we probably would reject it

Here are the reasons again a bit more detailled:

1. As already said we are in the application layer. SSL termination shouldn't be part of this layer. We are working at the Single Responsibility Principle. If we add SSL to the image itself we are also responsible for support and maintenance of the SSL termination of Wavelog. This is beyond our scope. 
2. Using reverse proxies for SSL termination is industrial standard. You have much more possibilities by using a reverse proxy the using SSL internally. HTTPS is mandatory, yes, but only if you leave your network. Even in a zero trust environment it's absolutely okay to use unencrypted traffic on your local machine (local reverse proxy talks to http of wavelog container). This produces much less overhead and is safe enough (with all the other security concerns you have to think about)
3. Support, because people would mount certs and paths wrong, don't understand the difference between valid and self-signed, and so on.. 

All these things are way beyond our scope. We are the maintainer of the software and we provide an easy way to deploy it. This way we have more time to develop database performance, statistics, new features, fixing bug, and so on. Everything which is not directly related to Wavelogs Code itself (and SSL is not) just reduces the amount of time we can continue working on this project. 

Therefore no. While we are discussing here I also talked to parts of the team and we are on the same page. I again fully agree that https is absolutely mandatory when exposing things to the outside world. If you want to use the "Keep me logged in" feature, yes, you need https. But still its super easy to just add a nginx or traefik to the docker compose file. 

I hope you understand that.

Cheers