wavelog/wavelog #2208: All cron jobs are failing

blhoward2, 2025-08-02:
**Describe the bug**
Last week, wavelog properly executed cron jobs uploading and downloading from LOTW, etc. Last night I realized they are now all failing when run from cron manager; however, they all work properly when manually synced through the Third Party Services menu.

I tried updating to the latest docker image, with the same result. 

With logging set to 4, no errors are logged:
```
root@cde81feccc8d:/var/www/html/application/logs# tail -f log*
INFO - 2025-08-02 16:18:01 --> Model "User_options_model" initialized
INFO - 2025-08-02 16:18:01 --> Model "Logbooks_model" initialized
INFO - 2025-08-02 16:18:01 --> Model "Stations" initialized
INFO - 2025-08-02 16:18:01 --> Model "Oqrs_model" initialized
INFO - 2025-08-02 16:18:01 --> Controller Class Initialized
INFO - 2025-08-02 16:18:01 --> Model "Cron_model" initialized
DEBUG - 2025-08-02 16:18:01 --> CRON: clublog_download is due and will be executed.
DEBUG - 2025-08-02 16:18:01 --> CRON: qrz_download is due and will be executed.
INFO - 2025-08-02 16:18:01 --> Final output sent to browser
INFO - 2025-08-02 16:18:01 --> Total execution time: 0.1548
```

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Latest docker image
- PHP:             8.3.23
- MySQL/MariaDB:    11.3.2-MariaDB-1:11.3.2+maria~ubu2204
- Wavelog:          2.0.7 

**Steps to Reproduce**
1. Schedule a cron job to run
2. Job fails and is marked as failed in UI
3. No error messages logged
4. Manually running a sync for each does work and new QSOs are downloaded, etc.

**Expected Behavior**
Cron jobs should run as scheduled.

**Screenshots**

[image]

**Desktop Information:**
- OS:               NixOS
- Browser:          Firefox
- Version:          140.0.2

**Additional Context**
Everything worked last week before I went on vacation. No changes, and it didn't work when I got home.

blhoward2, 2025-08-02:
Also, to be clear it is literally ALL cron jobs failing. I had every single one enabled and they all failed, from the version update check to QRZ, LOTW, etc.

blhoward2, 2025-08-03:
Thoughts on how to diagnose it? It's a docker image and everything on the host machine is working fine.

HB9HIL, 2025-08-03:
> Thoughts on how to diagnose it? It's a docker image and everything on the host machine is working fine. The jobs do trigger, they just fail.

Did you made any changes inside the docker container?

HB9HIL, 2025-08-03:
If a job fails you will in the log only see that it fails. Simplified there are two main reasons why cronjobs fail in Wavelog:

1. The machine does not have any internet access or can't reach out to the targeted services (LoTW, Clublog, etc). This can happen due to blocked outbound traffic or misconfigured DNS.

2. You made any changes inside the docker container. The docker installation comes with a preconfigured cronmanager. No changes should ever made inside the container (this is why it's a container). The cron inside the container calls wavelog on http://localhost which is absolutely fine. If you made any changes kill the container, repull the image and restart it. 

I can't see any other reason at the moment why ALL jobs should fail in the same way. 

Reference in code: https://github.com/wavelog/wavelog/blob/master/application/controllers/Cron.php#L114-L116

Are you running the docker image as we recommend it with docker compose ([Reference in Wiki](https://github.com/wavelog/wavelog/wiki/Installation-via-Docker#docker-composeyaml))?

Or are you running this NixOS Package [(Reference)](https://github.com/NixOS/nixpkgs/pull/371658)

blhoward2, 2025-08-03:
I made no changes inside the docker, and I pulled a new image and recreated to move to 2.07 with the same result. I also dropped a shell into the docker container and confirmed that I can ping the internet.

HB9HIL, 2025-08-03:
Currently testing our latest docker image... There is indeed something fishy. But can't tell you what atm. 

@int2001 There is an issue. curl in cron controller fails for no reason. $crun is empty. We need to check

HB9HIL, 2025-08-03:
https://github.com/wavelog/wavelog/wiki/Recommended-Cron-Jobs-and-Cronmanager#important-tip-for-docker-installs

This solves your issue

HB9HIL, 2025-08-03:
It's not a bug, but indeed a bad thing.. This maybe needs some more automation as I get caught by this trap aswell (and I wrote this shit :-D)

blhoward2, 2025-08-03:
Testing. Is there a reason it just doesn't default to that when the docker environment variable is set?

HB9HIL, 2025-08-03:
Thanks @blhoward2 for poking me here. PR 2214 solves this finally .. I hope so

blhoward2, 2025-08-03:
Glad it was an easy fix!