wavelog/wavelog #2217: Issue with Uploading LoTW p12 Certificate and Login

mc-doge666, 2025-08-04:
**Describe the bug**
When uploading my LoTW p12 certificate, I receive the error:
```
error:10000080:BIO routines::no such file
```
Additionally, I am unable to log in to LoTW from the account information page. The error displayed is:

```
Error: LoTW is currently unavailable. Please try again later.
```

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS: Windows11
- PHP:  8.2.29
- MySQL/MariaDB: 9.1.0
- Wavelog:  2.0.7

**Steps to Reproduce**

1. Attempt to upload the LoTW p12 certificate.
2. See the error message error:10000080:BIO routines::no such file.
3. Go to the 'Account Information' page.
4. Try logging in to LoTW and see the error message Error: LoTW is currently unavailable. Please try again later.

**Expected Behavior**
I expected to be able to upload the LoTW certificate without errors and log in successfully to LoTW.

**Screenshots**

[image]
[image]
[image]

**Desktop Information:**
- OS:  Windows11
- Browser:    MS Edge
- Version:  140.0.3456.0

****I took the screenshots using the Chinese localization version and then translated them into English using a translation tool, so some words may differ from the English version.****

AndreasK79, 2025-08-05:
Hi, perhaps this is not documented. To be able to read and upload certificates, you need to install openssl in Windows.
I don't know if openssl also affects the login test though.

mc-doge666, 2025-08-05:
It changed the error message again. error:07000072:configuration file routines::no such file

AndreasK79, 2025-08-28:
Did you manage to solve this one?

mc-doge666, 2025-11-11:
> > I'm not sure if the following issue stems from the problem described in this issue. After fixing the problem where P12 certificates couldn't be imported, LOTW still fails to log in as shown in the image, and it cannot sync any third-party information from the cloud, including logs. Therefore, I reopened the issue.
> 
> This looks like your Wavelog instance cannot connect to LoTW. Maybe some firewall or filter issue?

I temporarily disabled all firewalls on the system, but I still cannot connect to LOTW and all other third-party services.