wavelog/wavelog #2219: Confusion about multiple station profiles with same callsign

dl8dtl, 2025-08-04:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**
When looking up a QSO in the simple view, the "station info" occasionally reports wrong data.

**Describe the bug**
I have been multiple times in OZ, so I have multiple OZ/DL8DTL station profiles.
It appears that the station info reported in the simple log view (after a search for a particular QSO) always reports the same location from one of the three station profiles.
Using the advanced view, restricting it to the correct profile makes it clear it has been entered correctly into the log. Using the "station  info" there also reports it correctly then.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               FreeBSD 13.4
- PHP:              8.2
- MySQL/MariaDB:    9.0.1
- Wavelog:          2.0.7

**Steps to Reproduce**
1. create multiple station profiles with same call
2. log QSOs to them
3. go to simple log view
4. search for one callsign
5. look at "station info" shown

**Expected Behavior**
The correction station info should be there.

**Screenshots**
See attachments

**Desktop Information:**
- OS:               MacOS
- Browser:          Firefox
- Version:          134.0.2

HB9HIL, 2025-08-04:
[image]

Seems like you have the same QSO logged twice in two different station locations.

dl8dtl, 2025-08-04:
Well, might be because of (see discussions) the confusion between "location" and "logbook" when importing my old log data. Anyway, it would be good to have an indication about the station profile (rather than the callsign only) in the display.

HB9HIL, 2025-08-05:
Well in this case I'd say we move this issue to discussions (because in fact it is no issue/bug) and you can create another feature request in [Ideas](https://github.com/wavelog/wavelog/discussions/categories/ideas). Then we work out together what would be a good way to improve the situation here.