wavelog/wavelog #2265: Add support for QSO querying and editing to the API

mathuin, 2025-08-25:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
It is not currently possible to query or edit QSOs from the API.

**Why is This Feature Important?**
There are folks out there who participate in POTA as hunters -- the ones that respond to "CQ POTA".  Doing so via FT8 effectively strips the POTA from the exchange, so when the QSO ends up in Wavelog via the API there is no indication that it represents a POTA contact.  Instead the user is required to manually edit the QSO to supply the POTA_REF value.  Wouldn't it be nice to do that via batch?

The POTA website has a "hunters log" feature which allows users to browse the contacts that have been reported.  There are tools which convert this log into ADIF.  It would be nice to iterate over my hunter's log, query the Wavelog API for those QSOs, and push an edit with an updated POTA_REF field value.

The specific instance of this issue, with POTA, could be solved with a Wavelog internal cron job and the user could store their credentials not unlike LotW/eqsl/QRZ but that will introduce dependencies on the POTA site -- having a general solution lets users build their own tools, something which has been successful so far.

Multiple changes to the API would be required for implementation:
* an API call to query QSOs based on callsign and timerange that would return the Wavelog QSO index and the associated ADIF record
* an API call to edit QSOs based on QSO index and updated ADIF record

**Is This Feature Personal or Beneficial to Others as Well?**
Without this feature, Wavelog users who leverage the API to log their contacts will have to manually adjust individual contacts to reflect their xOTA participation, which is a drag.

phl0, 2025-08-28:
> The POTA website has a "hunters log" feature which allows users to browse the contacts that have been reported. There are tools which convert this log into ADIF. It would be nice to iterate over my [...]

Can you please name such a tool which converts the hunter log to ADIF? Might make an import on that.

mathuin, 2025-08-28:
> Can you please name such a tool which converts the hunter log to ADIF? Might make an import on that.

https://github.com/adamfast/pota-hunter-log-adif This was the first one that I found in a quick search, but there could be others.  I haven’t tried just importing the ADIF because I expect Wavelog to see the QSOs as duplicates and not replace the old ones with the new ones — if it _does_ replace them then I’m happy to manually generate ADIFs every so often and import them to get the desired result.

phl0, 2025-09-06:
> https://github.com/adamfast/pota-hunter-log-adif This was the first one that I found in a quick search, but there could be others. I haven’t tried just importing the ADIF because I expect Wavelog to [...]

I made a POTA ref import funtion in https://github.com/wavelog/wavelog/pull/2284. This should allow importing reference information from a hunter ADIF log. Would be nice if you could give it a try.