wavelog/wavelog #2294: Attempting QRZ.com upload when disabled does not show error message.

RhinoLance, 2025-09-08:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
In a "Station Location", under the QRZ.com section, if "Logbook Upload" is disabled, attempted sync shows a message "Undefined".

When disabled, the [qrzlogbook.js] return object has only a _data.status_ property, set to "QRZ Disabled for station1".  However the code is trying to evaluate _data.errormessages_ resulting in an undefined error.  (https://github.com/wavelog/wavelog/blob/c16b880bbdc754aff182983aee565be21cab82ac/assets/js/sections/qrzlogbook.js#L30)

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               ubuntu0.20.04.3
- PHP:              8.3.14
- MySQL/MariaDB:    8.0.37
- Wavelog:         2.1

**Steps to Reproduce**
1. Go to Station Setup -> User Station Location -> Edit
2. Set Logbook Upload to Disabled
3. Save
4. Go to Third-Party Services -> QRZ Logbook
5. Click Upload
6. Observe the red error box with the text "undefined" within.

**Expected Behavior**
Display the actual error message of "QRZ Disabled for station1"

**Screenshots**

[image]

[image]

[image]

**Desktop Information:**
- OS:               Windows 11
- Browser:       Edge
- Version:        139.0.3405.125

phl0, 2025-09-08:
Created a patch to deal with this: https://github.com/wavelog/wavelog/pull/2295.

phl0, 2025-09-08:
Patch has been mergerd. Please test and report.