wavelog/wavelog #2312: error with editing QSO after functioning normally

zcrowman, 2025-09-14:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
I'm going through my QSOs to update missing ITU zone information for that award. I had editing maybe 20 using the advanced logbook, sorted by ITU zone. During the editing of one QSO, the edit screen froze and wouldn't pull in information from Callbook. Clicking on the tabs selected the tabs but the screen didn't change. I clicked the X to close the window and when I did that the QSO populated the missing information from Callbook.  However, subsequent edits failed when clicking on the Edit QSO button with a screen that says: 
"An Error Was Encountered
Unable to load the requested file: qso/edit.php"

However, if I click on a call sign in the dashboard or overview, editing is still possible without error. Seems to only be associated with the Advanced log. 

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:     Ubuntu          [e.g. Debian 12]
- PHP:       8.3       [e.g. 8.2]
- MySQL/MariaDB:   MyQSL - version unknown [e.g. 10.11]
- Wavelog:   2.1       [e.g. 1.0] 

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
Describe what you expected to happen.

**Screenshots**
Include screenshots if applicable to illustrate the issue.

**Desktop Information:**
- OS:      MacOS Sonoma         [e.g., iOS]
- Browser:    Firefox      [e.g., Chrome, Safari, Firefox]
- Version:   142.0.1       [e.g., 22]

**Smartphone Information:**
- Device:           [e.g., iPhone6]
- OS:               [e.g., iOS8.1]
- Browser:          [e.g., stock browser, safari]
- Version:          [e.g., 22]

**Additional Context**
Provide any other relevant context about the problem.

zcrowman, 2025-09-15:
I'm following up on this issue with more information. This morning I reinstalled Wavelog on my webserver and that fixed the error message. So I carried on going through QSO's missing ITU zone information. I opened up the advanced log, sorted by ITU zone and then started editing. The first 5 or 6 worked fine. During editing the QSO, I hit the Update From Callbook button and it filled in some missing grid squares as well as ITU zone.  However, the more QSOs I edited the slower things got until it got to the point where opening the QSO for editing happened in slow motion and then it froze. Clicking on Update From Callbook didn't appear to populate the information and clicking on the Award tab did nothing. It is like the server ran out of resources and nothing would happen. I could Save the QSO but from then on I'm stuck in this slow motion dance where nothing seems to happen. However, if I switch to the Overview or Dashboard view and edit a QSO, it works normal with none of this freezing and slowdown.

iu2frl, 2025-09-16:
Any other diagnostic tool you can use?

AndreasK79, 2025-09-17:
@zcrowman I can't reproduce this one, but I suspect it might be with the datatable component we use for the main table in the advanced logbook. How many QSOs are you showing at the same time? If you are showing more than 250, try to set it to 250 and see if the problem is still there.

I am not sure we can do much about this, as I have seen slowdowns on my local Windows machine where I develop as well. That is the main reason I capped number of results to 5000.

iu2frl, 2025-09-17:
@zcrowman do you have the system specs for that VPS? Seems like the database is struggling

AndreasK79, 2025-09-17:
You can search for empty itu zones using the filter. You can also mark all those and update from the callbook, instead of doing one qso at a time.

zcrowman, 2025-09-17:
Obviously not using Wavelog to its full potential. Thanks for all your help.