wavelog/wavelog #2321: [FEATURE REQUEST] Print Logbook as PDF/Paper

Builderhummel, 2025-09-16:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
Describe as precisely as possible what you believe is missing. Also, explain how we could achieve the inclusion of this feature.
The feature should allow to print the QSO-Log in landscape and portrait using multiple filters (e.g. print every 15m CW QSO from 01.01.25-01.06.25 containing the following fields: "callsign, locator, rapport,..."). This could be achieved by adding a new button in "/logbookadvanced" called "Print to PDF"/"Export to PDF".

**Why is This Feature Important?**
Not every feature holds equal importance. Clarify why you think your suggested feature is crucial and how it enhances Wavelog.
This feature helps to create a fast paper backup for the logs. This is especially important for club stations, which are trying to switch to electronic logs, but still want to keep a paper copy. Moreover, annotating on paper is preferred by some hams. Also, it adds access when no electricity is present (e.g. to run a computer to access wavelog)

**Is This Feature Personal or Beneficial to Others as Well?**
If the feature is solely for your benefit, and not beneficial to the broader Wavelog community, we may not prioritize or respond to your request. Ensure that your proposed feature contributes to the entire Wavelog user base.
Every user has the benefit of being able to print out the log as paper backup.

Builderhummel, 2025-09-16:
Hi @HB9HIL,

in my understanding, the software is a way to store, manage and represent QSO Logs. A printout is just another form of representation.

Of course, you can write another tool to printout ADIF files (or simply consume the API, etc.), but then you would need, for example, an additional data storage to keep track which QSO was already printed as well as separate host for the new application (e.g. the same computer where the log was entered).

But what happens if you want to have quick access to the printout file on any device (e.g. your smartphone or printer terminal) (i.e. one of many reasons to choose a web app in the first place)?

In my opinion, such a feature provides important functionality, convenience and should not be too complicated to implement.

phl0, 2025-09-16:
> Of course, you can write another tool to printout ADIF files (or simply consume the API, etc.), but then you would need, for example, an additional data storage to keep track which QSO was already [...]

Where would you store the information which QSOs were printed if the print function was implemented? There are no (ADIF) fields to store such information.