wavelog/wavelog #2324: Is it possible to display a US stations state when looking up a call?

brstiefel1, 2025-09-16:
If I search for a callsign or try to log a station in the US, it will display the city but not the state of the callsign. Is there a way to change this? If not, can this be turned into a feature request?

HB9HIL, 2025-09-17:
@brstiefel1 

I use qrz.com as callbook source. I also have the XML subscription (the cheapest one). If I log a US Callsign and this operator has properly set up his QRZ.com Account, the state will be prefilled (Tested with N7DZ)

[image]

[image]

It may be possible that you need at least the XML Subscription at QRZ.com to get the state in the query. Unfortunately this is not something we can solve within Wavelog. If you already have the XML subscription and still got no state for another callsign, the reason may be that this user did not entered this information in his account. There is NO geo-auto-magic autofill of the state based on the position/grid, because simply way to complex.

HB9HIL, 2025-09-17:
So from my point of view this is already implemented. Converting this issue to a discussion.