wavelog/wavelog #2331: Could not migratde atabase. Ajax crashed. Statu 500 Status Text: s:Internal Server Error

mimihzs, 2025-09-20:
Wavelog Installer Debug Log
-------
Log Location: ../application/logs/installer_debug.log

DEBUG - 2025-09-20 21:03:33 --> Start writing config.php
INFO - 2025-09-20 21:03:34 --> Data validation passed.
INFO - 2025-09-20 21:03:34 --> CI_ENV is not set. Using default config.php config path.
INFO - 2025-09-20 21:03:34 --> config.php template file read successfully.
INFO - 2025-09-20 21:03:34 --> Config.php file prepared successfully. Writing to file...
INFO - 2025-09-20 21:03:34 --> config.php file written successfully.
DEBUG - 2025-09-20 21:03:34 --> File: config.php successfully written
DEBUG - 2025-09-20 21:03:34 --> Start writing database.php
INFO - 2025-09-20 21:03:35 --> Data validation passed.
INFO - 2025-09-20 21:03:35 --> CI_ENV is not set. Using default database.php config path.
INFO - 2025-09-20 21:03:35 --> database.php template file read successfully.
INFO - 2025-09-20 21:03:35 --> Database config file prepared successfully. Writing to file...
INFO - 2025-09-20 21:03:35 --> database.php file written successfully.
DEBUG - 2025-09-20 21:03:35 --> File: database.php successfully written
DEBUG - 2025-09-20 21:03:35 --> Start creating database structure with assets/install.sql
INFO - 2025-09-20 21:03:35 --> Data validation passed.
INFO - 2025-09-20 21:03:35 --> SQL queries prepared successfully. Writing to database...
INFO - 2025-09-20 21:03:36 --> Database tables created successfully.
DEBUG - 2025-09-20 21:03:36 --> Tables successfully created
DEBUG - 2025-09-20 21:03:36 --> Start migrating database to the newest version.
ERROR - 2025-09-20 21:03:36 --> Could not migratde atabase. Ajax crashed. Statu 500 Status Text: s:Internal Server Error

HB9HIL, 2025-09-20:
It would be really really helpful if you can gove a few more informations about your setup, how to reproduce. What do you think shall we do now with this logfile..

Sorry. Try again

closed 2025-09-20 by HB9HIL

wohenbuguai, 2025-11-11:
我也有相同问题，使用的是aapanel，php8.2